CDFER/JLCPCB-Kicad-Library #10: Feature Request: Add support for Extended parts from LCSC

DeflateAwning, 2024-12-10:
This project is super awesome!

From the README, it looks like only Basic parts are supported:

> All components are from JLCPCB's basic and preferred parts list, ensuring no extra setup costs.

I'm wondering if you've considered adding support for the entirety of the LCSC parts inventory (called "Extended" parts in JLCPCB)? It would be very helpful to be able to access all parts in their inventory via this project. Might make sense to have it released in a separate repository or something though, as it is >100x as large as the Basic parts iirc.

CDFER, 2024-12-10:
See my comment here, supporting 300,000+ components isn't really feasable rn but if you have any particulalry useful parts I'm up for adding them to the "JLCPCB-Extended" sublibrary that I have.

https://github.com/CDFER/JLCPCB-Kicad-Library/issues/8#issuecomment-2526406619

DeflateAwning, 2024-12-10:
Ah neat, all very fair! 

Sounds like an opportunity for on-demand generation in a KiCad extension or something (in the distant future ofc).

closed 2024-12-12 by CDFER