CDFER/JLCPCB-Kicad-Library #13: Feature Request: Make this library installable via the KiCad extension manager

DeflateAwning, 2024-12-14:


CDFER, 2024-12-14:
This is something I've been thinking about, but I don't think KiCad would allow it in the normal package repository (https://gitlab.com/kicad/addons/repository) due to some of the files being pulled from EasyEDA and the copyright/API access issues that come with that, which KiCad is strict about (https://dev-docs.kicad.org/en/addons/). That being said, I could create my own third-party repository or ask Bouni to join his (https://github.com/Bouni/bouni-kicad-repository), which he uses for his tools here: https://github.com/bouni/kicad-jlcpcb-tools

Being able to press a button in KiCad to update the library would be very handy, so this is pretty high on my to-do list when I get some time.

DeflateAwning, 2024-12-14:
I figure it's worth a shot submitting it to the official KiCad one? And just see what they think? I don't think JLCPCB/LCSC/EasyEDA has much of a history of litigation. And this project doesn't explicity re-distribute; it just uses it as an "inspirational data source" of sorts.

CDFER, 2025-01-12:
Installing via the package manager has been added in V3 that I have just released. For now I am using my own package repository hosted here: https://github.com/CDFER/cd_fer-kicad-repository but I'll be talking to KiCad and Bouni soon :)

closed 2025-01-12 by CDFER