CDFER/JLCPCB-Kicad-Library #18: 0805 Footprint Errors

frohro, 2025-02-06:
Hi,
The 0805 footprints you are using have a "malformed courtyard."  

[image]

Thanks for fixing all these things!

Rob

CDFER, 2025-02-07:
Thanks for spotting this, it really helps to get rid of those rough edges that I have missed :)

I've gone through and fixed all the DRC errors I can find in the footprints here: https://github.com/CDFER/JLCPCB-Kicad-Library/commit/b3a3518cb78484469e243446e26005330055260b

It'll be out in the release later today.

One thing I still need to work on is standardizing how the courtyard is sized across the footprints (not all that useful to most people but pretty handy if you use auto-placement in tools like TopoR)

frohro, 2025-02-07:
Thanks Chris!

frohro, 2025-02-09:
They appear good now Chris!

closed 2025-02-09 by frohro