CDFER/JLCPCB-Kicad-Library #20: C5446 Part symbol has Vin and Vout pins swapped.

vsch, 2025-03-29:
The part has wrong pinout on the symbol. It is an LDO 3.3V regulator and the symbol has Vin and Vout swapped. From the datasheet (https://jlcpcb.com/api/file/downloadByFileSystemAccessId/8579707861620228096):
[image]

JLCPCB part page <https://jlcpcb.com/partdetail/TorexSemicon-XC6206P332MRG/C5446> shows the right EasyEDA symbol, 
[image]

however JLCPCB-Kicad-Library and importing from EasyEDA using the `easyeda2kicad` tools, results in the wrong pinout also.

JLCPCB-Kicad-Library:

[image]

easyeda2kicad:

[image]

I can confirm this from 10 PCBs that I ordered and the 3.3V supply was not working. Swapping the pins corrected the problem. I now have to do this for the other 9 PCBs.

CDFER, 2025-03-31:
Thank you for reporting this issue—and especially for including detailed references (datasheet, JLCPCB links, and screenshots). You’re absolutely right: the symbol in both the JLCPCB-Kicad-Library and easyeda2kicad output incorrectly swaps Vin and Vout. I’ve verified this against the datasheet and confirmed the discrepancy.

Next Steps:
* Fix the Library: I’ve corrected the symbol in the library and it will be pushed out in the update later today.
* I've gone through the JLCPCB-Power Library by hand and checked all the symbols (I found that C3113 was wroing too)
* I have put together a quick bit of code to find other errors using deepseek R1 and I will pulish this soon along with any fixes that I need to made once it has finished processing (it takes an hour to and it's running rn)
* When I have time (or if someone else has the time to do it before me) we should go through the other libraries based on easyeda2kicad these are:
```
 -Analog
 -Connectors_Buttons
 -Crystals
 -Diode-Packages
 -ICs
 -Inductors
 -Interface
 -MCUs
 -Memory
 -Transistor-Packages
```

I’m sorry this caused issues with your PCBs—I appreciate you taking the time to flag it so others won’t run into the same problem. Going forward I'm going to go through and double check all the footprints from easyeda2kicad with thier respective data sheets. Let me know if you spot anything else :)

CDFER, 2025-03-31:
Issue for easyeda2kicad.py:  uPesy/easyeda2kicad.py/issues/156
JLC2KiCad_lib also has the same issue too: TousstNicolas/JLC2KiCad_lib/issues/83

closed 2025-04-17 by CDFER