CDFER/JLCPCB-Kicad-Library #25: importing into fusion breaks with 3DPARTY relative hack

rbhun, 2025-04-27:
the kicad_mode files use this path:
(model "${KICAD8_3RD_PARTY}/3dmodels/com_github_CDFER_JLCPCB-Kicad-Library/JLCPCB.3dshapes/

this is not recognized by fusion360.

I have not found a way to set it so fusion 360 gets the 3D model linked.

CDFER, 2025-04-27:
Hmm, I just imported a into fusion 360 yesterday. I'm not sure I understand what you mean by gets the 3D model linked.

This is the way I do it:

1. Export STEP from KiCad

[image]

[image]

2. Upload to Fusion 360

[image]

Here are some renders I did with Fusion 360 yesterday

[image]

[image]

rbhun, 2025-04-28:
yes, the step files get into Fusion. Hoewever, the link is broken - so if I import any  symbols in this library, I'll end up with a symbol, with a placeholder 3D file, plus a 3d file on its own. The (model) link that is in the kicad_mod file is not followed by fusion - at least that is my experience. I looked around and it seems Fusion doesnt like that $KICAD_8_3RDPARTY variable, but even with straight folder structure I cannot get it to parse the 3d file - it just uploads it.

CDFER, 2025-04-28:
I'm still not sure what you are trying to do. To help investigate this issue, could you share some screenshots? 

1. A **step-by-step breakdown** of what you're doing 
2. What you **expected** to happen vs. what **actually** happened

closed 2025-06-01 by CDFER