CDFER/JLCPCB-Kicad-Library #26: DP9900 series of POE modules appear to be missing

jharvey, 2025-04-28:
There are a series of POE modules which would be great to have in the library. I guess extended parts are not the primary goal. It would be great to see these module added to the library? 
https://jlcpcb.com/partdetail/Sdapo-DP9900M5V/C5380106

jharvey, 2025-04-28:
If it helps, here's a symbol which works for the DP990X series, and a footprint which works for the DP9900M series.
-- The symbol and footprint were made by converting a EasyEDA file to KICAD, using online converter tools. 
-- 3d Step file was made by me using OnShape and MFG datasheet. I can make the DP9900LP series if there is interest. 
I release this under what ever open source license is preferred. 

[DP990.zip](https://github.com/user-attachments/files/19938473/DP990.zip)

CDFER, 2025-04-28:
Cool, I'll add it when I get some free time. I just need to convert it to KiCad 8 so we maintain bacwards compatability :)

jharvey, 2025-04-29:
In the next week or so I'll have a couple more components. Looks like an ETH jack and an ETH transformer. Any suggestion on things I can do with those components to help? I'll try doing it in KC8, then use that lib file in my KC9 project.

jharvey, 2025-04-30:
@CDFER 
I changed the issue description. Also please see attached, an ETH connector, an ETH transformer, and a POE module. These include symbol, footprint, and 3D files. I only made 3D of the DP9900M, all other files are imports from EasyEDA with light manipulation. 

[KC8.zip](https://github.com/user-attachments/files/19969921/KC8.zip)

jharvey, 2025-04-30:
Hmmm, how to convert the KC9 backwards into KC8. My KC9 symbols are much nicer then the KC8's imported from EasyEDA. I'll attempt to convert the KC9's to KC8's tomorrow.

jharvey, 2025-04-30:
I found a way to convert. In EasyEDA I imported a KC file, then I exported to Altium, then I opened that with KC8. If you open KC8 eeschema.exe manually, not via project launch pad, it allows you to open files. Then I did a bit of cleanup, and now have the files attached. Anyhow see below KC8 files converted from my KC9 files. 
[KC8_from_KC9.zip](https://github.com/user-attachments/files/19974745/KC8_from_KC9.zip)

jharvey, 2025-05-03:
Here's another one. It's a rectifier used in POE circuit. It was drafted in KC8. 

[MB6S.zip](https://github.com/user-attachments/files/20022542/MB6S.zip)

jharvey, 2025-05-03:
Another connector, this if for an OV2640 or OV5640 camera. This was exported from EasyEDA, and I'm relying on the pinout from here https://github.com/maxlab-io/tokay-lite-pcb
[ov2640-ov5640_CAMERA-CONNECTOR.zip](https://github.com/user-attachments/files/20023097/ov2640-ov5640_CAMERA-CONNECTOR.zip)

jharvey, 2025-05-03:
Also these symbols. Same as the 3.3V version, but for 2.8V and 1.5V. This was sourced from the existing CDFER library. 

[XC6206_symbols.zip](https://github.com/user-attachments/files/20023577/XC6206_symbols.zip)

jharvey, 2025-05-04:
Here's another one. It's a 7 segment display driver, Symbol / footprint exported from EasyEDA, 3D is KICAD original. 
[digital tube drivers_c2962580.zip](https://github.com/user-attachments/files/20026989/digital.tube.drivers_c2962580.zip)

jharvey, 2025-05-04:
A red 7 segment LED display 3 characters wide. The symbol, footprint and 3D exported from EasyEDA
[3x7-seg_LED-RED_C2913084.zip](https://github.com/user-attachments/files/20027095/3x7-seg_LED-RED_C2913084.zip)

@CDFER Should I  be doing this in separate issues, or is doing this all in 1 issue an OK place to relay the files?

jharvey, 2025-05-04:
This is an audio chip, Exported from EasyEDA, 3D is KICAD OEM. 
[MAX98357AETE+T.zip](https://github.com/user-attachments/files/20028516/MAX98357AETE%2BT.zip)

jharvey, 2025-05-06:
I found a problem with several of the symbols I posted above. Several need a rotation to prevent issues when sent to JLCPCB. I'll update those in the next couple of days. I'm also not happy with my inconsistent naming convention. I plan to change these 2 things and re-upload a new version of the symbols / footprints. 

I tried following the pull request guidelines, but I was not successful in the validation step of setting up VSCode.

jharvey, 2025-05-07:
@CDFER  I just took a stab at a pull request. I'm not a GH guru. Let me know if something is out of place. I think it should be good. I'd suggest double checking the 3D and the footprint are linked correctly. For some reason my PC has an issue with certs and I used the GH clone repo instead of the plugin manager repo. This may have potentially caused some linking issues, but I think I got it all proper. 
-- Files were made with KC8
-- Some files use KC 3D
-- Most files were exported from EasyEDA. I made a couple symbols with more detail. I noted those with KC. They are kind of duplicates of the EDA simple version. 
-- 3D was mostly exported from EaseEDA, lightly manipulated in onshape. I created one model from the MFG datasheet. 

I'll remove the above zip files and extra posts. I think those are just going to cause confusion at this point.