CDFER/JLCPCB-Kicad-Library #30: JLC part rotation issues for PCM_JLCPCB:SOIC-8_L5.3-W5.3-P1.27-LS8.0-BL

jharvey, 2025-05-12:
Hello,

When I place LCSC C97521it uses footprint PCM_JLCPCB:SOIC-8_L5.3-W5.3-P1.27-LS8.0-BL. Fab files were generated with plug in tool and uploaded to JLCPCB. Then I get a rotation issue as noted in this picture.

[image]

The 3D files in KICAD appear to be in good order. It's just a problem with JLC files.

[image]

I'm having this with a couple other packages as well. I plan to put in a new issue for each footprint. I'm assuming you would prefer individual issue trackers for each footprint.

My JLCPCB KiCad Library is Version 2025.05.12
Plugin tool "Fabrication Toolkit" https://github.com/sponsors/bennymeg version 5.0.1
KiCad V9.0.2 
Windows 10.

CDFER, 2025-06-01:
I've merged your two issues. Here is your other issue:

Hello,

When I place LCSC C15127 it uses footprint PCM_JLCPCB:Q_SOT-23. Fab files were generated via Plugin tool "fabrication toolkit" https://github.com/sponsors/bennymeg version 5.0.1 and uploaded to JLCPCB. When I do I get a rotation issue as noted in this picture. 

[image]

The 3D files in KICAD appear to be in good order. It's just a problem with JLC files. 

[image]

I'm having this with a couple other packages as well. I plan to put in a new issue for each footprint. I'm assuming you would prefer individual issue trackers for each footprint. 

My JLCPCB KiCad Library is Version 2025.05.12

CDFER, 2025-06-01:
Thanks for reporting this issue and including detailed examples :)

### **Root Cause**  
The rotation mismatch occurs because JLCPCB’s assembly process uses a different **coordinate system** and **rotation reference** compared to KiCad. While the 3D models in KiCad appear correct, some footprints in this library (especially those adapted from EasyEDA) have orientation that conflict with JLCPCB’s expectations. The Fabrication Toolkit plugin assumes native KiCad footprints, so non-standard rotations in modified/EasyEDA-derived footprints can cause misalignment in the generated CPL files.  

---

### **Solutions Under Consideration**  
1. **Rotation Mapping File**  
   - Create a `.csv` file to manually override rotations for problematic footprints (compatible with [kicad-jlcpcb-tools](https://github.com/Bouni/kicad-jlcpcb-tools)).  
   - **Pros**: Immediate fix for existing footprints.  
   - **Cons**: Requires users to import the file for every project.  

   [image]  

2. **Transition to Native KiCad Footprints**  
   - Gradually replace EasyEDA-derived footprints with KiCad-native ones to ensure compatibility.  
   - **Pros**: This already slowly happening to get away from large mesh in .step 3dfiles that easyEDA sometimes uses.
   - **Cons**: Time-intensive; some footprints (e.g., JLC-specific parts) will need to be recreated from scratch.  

---

### **Short-Term Workaround**  
- For urgent projects, manually adjust rotations in the CPL file or JLCPCB’s assembly interface. 
- Use **Bouni’s kicad-jlcpcb-tools** with a custom rotation file. 

---

### **Next Steps**  
- I’ll prioritize fixing the **SOIC-8** and **SOT-23** footprints you flagged. 
- If you encounter other problematic parts, just add them to this issue