CDFER/JLCPCB-Kicad-Library #34: JLCPCB Changes Parts from Basic to Extended

frohro, 2025-06-24:
Hi,
I updated just last night, but now C25796 from your basic parts library is an Extended part.

[image]

CDFER, 2025-06-24:
The thumbs up Extended parts are the prefered parts and cost the same as the basic parts.

Yes JLCPCB have made it quite confusing (I think this is on purpose to charge people more).

frohro, 2025-06-24:
I doubt it is for that reason.  They have to load Extended parts on the pick and place machines.  I suspect they got a better price on some new parts that they decided to leave on the pick and place machines instead of the old ones.  :-)  Their site is still confusing.  That is why we need plugins like yours.  :-)

closed 2025-06-24 by CDFER